PaulStoffregen/AltSoftSerial #12: known_timers.h and known_boards.h not here on Github

GreyGnome, 2015-05-19:
Hi,
I notice that in config there seem to be a couple of files that are in the Zip but not here on Github. Not sure if that's correct or not but I thought I'd point it out because I had a bit of trouble finding the definition of CAPTURE_INTERRUPT, for example.

Thanks.

PaulStoffregen, 2015-05-20:
It's defined here.

https://github.com/PaulStoffregen/AltSoftSerial/blob/master/config/AltSoftSerial_Timers.h#L45

Are you sure there's something missing from github?  Can you be more specific?

GreyGnome, 2015-05-21:
Ah, I see. I thought there was something missing because I did the
following:

   - Wondered how COMPARE_A_INTERRUPT was defined
   - Did a Google search (COMPARE_A_INTERRUPT altsoftserial), first result
   was with a site that pointed me to a site that had config/known_timers.h
   (->
   https://github.com/RedBearLab/BLE_HCI/blob/master/arduino/libraries/AltSoftSerial/config/known_timers.h
   )
   - Browsed to
   https://github.com/PaulStoffregen/AltSoftSerial/tree/master/config, did
   not see known_timers.h . Wondered where it came from.
   - Downloaded library zip file, it had the file in it under config.
   - Posted bug report.

I was confused because I found a file in the zip file under config that was
not on the Github site.

PaulStoffregen, 2015-05-21:
Oh, yeah, the zip file is probably old by now.

On my to-do list is removing ALL the zip files from the website and replacing them with links to these github repositories, which are the most up-to-date copies.

closed 2016-02-06 by PaulStoffregen