PaulStoffregen/AltSoftSerial #13: Wrong timing on first start bit?

DougF42, 2015-06-05:
I was looking at "writeByte",  at line 124:
      SET_COMPARE_A(GET_TIMER_COUNT() + 16);

If I am reading the code right, this is setting the length of the start bit for the first byte. 
Instead of a constant of '16', I think it should  be set to "ticks_per_bit" ?

matthijskooijman, 2015-09-27:
No, this +16 just means "run the ISR ASAP to begin the start bit". I believe this prevents duplicating code between the ISR and `writeByte`. The ISR has code to set the line state and, depending on the data to be sent, schedule a compare match to toggle the line state, which is fairly complicated, but works the same for the startbit and the databits.

I believe "16" here is chosen as a fairly short interval, that is certainly enough to complete the read-modify-write cycle that `SET_COMPARE_A(GET_TIMER_COUNT() + 16);` is. So, AFAICS the code is correct (also logic traces show the start bits have the same value).

PaulStoffregen, 2015-09-27:
Ok, I just did a couple quick tests, with this code

    #include <AltSoftSerial.h>
    
    AltSoftSerial myser;
    
    void setup() {
      myser.begin(9600);
    }
    
    void loop() {
      myser.write(0xA5);
      myser.write((uint8_t)0);
      delay(100);
    }

On Teensy 3.2:

[image]

On Teensy 2.0:

[image]

I believe it's safe to conclude the start bit is fine.

closed 2015-09-27 by PaulStoffregen