PaulStoffregen/AltSoftSerial #17: ATtiny 85?

hoonkai, 2015-09-13:
Hi Paul,

This is an amazing piece of work. :)

Just wondering: are there feasible ways to massage AltSoftSerial to work on an ATtiny85? I see that about half a dozen boards are formally supported, but could something be done to make it support the attiny?

Cheers

PaulStoffregen, 2015-09-13:
Ask Atmel to make a version with the right 16 bit timer hardware?

closed 2015-09-13 by PaulStoffregen

josephernest, 2019-09-18:
@PaulStoffregen Would there be a way to make AltSoftSerial with ATTiny45/85?
Maybe I thought there would be new alter2015, maybe not requiring 16 bit timer, or any other technique?
Thanks in advance!

PaulStoffregen, 2019-09-18:
16 bit timer is required

josephernest, 2019-09-18:
Thank you for your answer. So impossible to have anything else than SoftwareSerial (aka NewSoftSerial) on ATtiny45?

josephernest, 2019-09-19:
@PaulStoffregen if I take an ATTiny84, will your library work out-of-the-box with it with Arduino IDE (I use https://github.com/SpenceKonde/ATTinyCore)? The 84 has a 16 bit timer, so I'm thinking about getting one.

Congratulations again for your library!

PaulStoffregen, 2019-09-22:
> if I take an ATTiny84, will your library work out-of-the-box

Probably not.  I have not personally tested this chip.  If you read AltSoftSerial_Boards.h, so far nobody has contributed defines to support it.  Maybe you will create these, test to make sure everything works, then contribute so others can have this chip work out-of-the-box?

josephernest, 2019-09-22:
I am currently active on a few other (opensource too) projects, and if I have some time, I will definitely contribute. (For now, I just took an ATmega328p in order to have a real hardware UART, and everything is working great thanks to HardwareSerial).
I've also looked at NeoSWSerial (another software serial library), but 31250 baud (for MIDI) is not supported for 8 Mhz chips, unfortunately!