PaulStoffregen/AltSoftSerial #20: Issue with Teensy 3.2 and AltSoft at 9K6 speed

VMEU, 2015-11-26:
Hi there, 

I have a problem which I cannot solve here: I use AltSoft for receiving bytes form a 9K6 serial keyboard interface on a Teensy 3.2. It works well for values between 0 and 126. As soon as the sent byte value increases to 127 the Teensy 3.2 recognises numbers in random values. As soon as I receive bytes smaller than 127 it takes 2-3 bytes to get the right output again. 

Here is the code:

#include <AltSoftSerial.h>
AltSoftSerial keyin;
byte keyinData = 0;

void setup()
{
  keyin.begin (9600);    // Keyboard Console Serial Receive (Pin 20)
  Serial.begin(9600);
}

void loop()
{
     if (keyin.available()>0)
      {
        keyinData = keyin.read();
        Serial.println(keyinData);
      }
}

Is there something I do wrong? Thank you for brining light into my darkness :.-)

Cheers,
Alexander

PaulStoffregen, 2015-12-06:
I tested this today, but could not reproduce the problem.  Since I don't have your serial keyboard, I used Serial1 to transmit to AltSoftSerial.  I just put a Teensy 3.2 on a breadboard and connected a wire from pin 1 to pin 20.

Here's the code I tested.

    #include <AltSoftSerial.h>

    AltSoftSerial keyin;

    void setup() {
      keyin.begin (9600); // Keyboard Console Serial Receive (Pin 20)
      Serial.begin(9600);
      Serial1.begin(9600); // connect wire from pin 1 (TX1) to pin 20 (AltSoftSerial RX)
    }

    elapsedMillis msec = 0;

    void loop() {
      if (msec > 1000) {
        msec = 0;
        Serial1.write(127);
      }
      if (keyin.available() > 0) {
        byte keyinData = keyin.read();
        Serial.println(keyinData);
      }
    }

Can you please verify you're using the latest AltSoftSerial version?

The ZIP file on the web page was an old, out of date version.  I've removed it.  The copy in Teensyduino 1.26 and here on github is the latest code.

VMEU, 2015-12-06:
Dear Paul,

I have tested your code with the latest teensy-software, the latest altsoft-version and 4 different teensy 3.2s. Your code works fine. 

But...
..as soon as I change the value "127" to "129" the chaos begins. The first reading in the serial monitor is the correct "129" and then it goes on with "224" (in my case here).

Alternatively I have tested your code with Serial3 (instead of altsoft) and values above "128" and here it works great.

Thank you,
Alexander

PS.: I really appreciate the things you are doing. Thank you for that!

VMEU, 2015-12-21:
Hi Paul,

I wonder if you could reproduce the issue with your teensies when sending values above 129?

Thank´s 
Alexander

PaulStoffregen, 2015-12-21:
Before I look at this *again*, I want you to verify a few things on your end.

First, make sure you are using the latest Arduino 1.6.7 and Teensyduino 1.27-beta1.  Use Help > About to check which version you are running.  Please do not report issues here with anything other than the very latest version.  Testing with old versions only risks reporting previously-fixed issues, which is just a waste of everyone's time.

Next, use File > Preferences to turn on verbose info while compiling.  Then click Verify.  A *lot* of info will print.  Scroll up.  You'll see a summary that shows the full pathname for the AltSoftSerial library Arduino actually compiled.  You must check to make sure it's really using the latest copy.  If place a copy in Documents/Arduino/libraries, it will override the one from Teensyduino.  Before you ask me to do anything more, check that you're truly using the copy from Teensyduino 1.27-beta1.  If you previously installed this into your libraries folder, you may be using the same old copy with issues that were fixed long ago.

With those checked, use the exact code I posted above.  Click "Verify", and then tell me the exact summary which prints (the amount of memory used).  I also need to know which operating system you're using.

Please do these steps before asking me to look at this yet again!

VMEU, 2015-12-21:
Hi Paul,
sorry if I only said that I used the latest software. To be more precise I use:

Arduino: 1.6.7
AltSoftSerial: 1.3.0 (yes, it´s used if I look at "verbose")
Teensy SW: 1.27 beta 1 (tried it at 1.26, too - same effect)
Teensy 3.2; on all CPU speeds
Microsoft XP SP3

The summery is:
Sketch uses 17,932 bytes (6%) of program storage space. Maximum is 262,144 bytes.
Global variables use 4,816 bytes (7%) of dynamic memory, leaving 60,720 bytes for local variables. Maximum is 65,536 bytes.

I can confirm your code works great.

BUT: It does not work as soon as the sent value is changed from 127 to 129 and above. That is the issue. I need to receive values up to 180.

I am happy to get your advice.

Thank you for your time,
Alexander

FrankBoesing, 2016-02-03:
I can confirm this.
It is indepenend from the Baudrate, as soon as Bit 7 (the first) is set, it does not work.
But transfering 0 does not work too (output is 255). Tested with your (Paul's) Sketch  from above. An issue with the first databit-edge(?)

FrankBoesing, 2016-02-03:
sorry, NOT independend. 1200 BAUD : Output  255 for a sent 0, 9600 Baud: the firstreceived Byte is 1, all following 192.

FrankBoesing, 2016-02-03:
0 works for 14400 and above, values >127 still not.

FrankBoesing, 2016-02-03:
One Bug is: for slow baudrates we have overflows on the 16 Bits values. Fix: Use a prescaler. This fixes the "0"-receiving problem. However, the problem when bit 7=1 still exists.

VMEU, 2016-02-03:
Frank, thank you for your confirmation. At least I am not the only one and I appreciate all your guys´ help. 

Cheers,
Alexander

PaulStoffregen, 2016-02-05:
I'm digging into this today.  The problem with values > 128 on Teensy 3.x seems to be a bug in setting compare B.  The ISR(COMPARE_B_INTERRUPT) isn't happening at the right time, or at all.  Still not sure why.....

PaulStoffregen, 2016-02-07:
I've fixed several subtle bugs and added support for slower baud rates.

Please give the latest code a try and let me know if it resolves this issue for you?

VMEU, 2016-02-11:
Great - thank you for your time, work and expertise. I am sorry that due an extended business trip I am not able to test it now. I´ll come back to you ASAP then.

PaulStoffregen, 2016-02-17:
I'm pretty sure the recent fixes solve this problem, so I'm closing this issue.

If there's still a problem, please open a new issue.  Remember to always post code & hardware details necessary to reproduce the problem.

closed 2016-02-17 by PaulStoffregen