PaulStoffregen/AltSoftSerial #26: Improved prescale

DrDiettrich, 2016-05-21:
Thanks for your fine code, both easily understandable and efficient :-)
Here is a suggestion for improved handling of the prescaler, better readable and easier to adopt to various prescaling factors. It also allows to report/flag invalid baudrates.
[prescale.zip](https://github.com/PaulStoffregen/AltSoftSerial/files/275793/prescale.zip)

PaulStoffregen, 2016-05-24:
From the zip file: 

    bool prescale(uint32_t cycles_per_bit)
    {
        ticks_per_bit = cycles_per_bit;
        if (ticks_per_bit < MAX_COUNTS_PER_BIT) {
                //Serial.printf("cycles_per_bit = %d\n", cycles_per_bit);
                CONFIG_TIMER_NOPRESCALE();
                return true;
        }
        ticks_per_bit = cycles_per_bit / 8;
        if (ticks_per_bit < MAX_COUNTS_PER_BIT) {
                //Serial.printf("cycles_per_bit/8 = %d\n", cycles_per_bit);
                CONFIG_TIMER_PRESCALE_8();
                return true;
        }
    #if defined(CONFIG_TIMER_PRESCALE_128)
        ticks_per_bit = cycles_per_bit / 128;
        if (ticks_per_bit < MAX_COUNTS_PER_BIT) {
                //Serial.printf("cycles_per_bit/64 = %d\n", cycles_per_bit);
                CONFIG_TIMER_PRESCALE_128();
                return true;
        }
    #endif
    #if defined(CONFIG_TIMER_PRESCALE_256)
        ticks_per_bit = cycles_per_bit / 256;
        if (ticks_per_bit < MAX_COUNTS_PER_BIT) {
                //Serial.printf("cycles_per_bit/256 = %d\n", cycles_per_bit);
                CONFIG_TIMER_PRESCALE_256();
                return true;
        }
    #endif
    #if defined(CONFIG_TIMER_PRESCALE_1024)
        ticks_per_bit = cycles_per_bit / 1024;
        if (ticks_per_bit < MAX_COUNTS_PER_BIT) {
                //Serial.printf("cycles_per_bit/1024 = %d\n", cycles_per_bit);
                CONFIG_TIMER_PRESCALE_1024();
                return true;
        }
    #endif
        AltSoftSerial::timing_error = true;
        //Serial.println("baudrate too low");
        return false; //baudrate too low
    }

DrDiettrich, 2016-05-24:
I've just created a fork and want to add complete and test the modifications there.
But perhaps you already see what I want to change, and how you can incorporate it into your library- your comment is not very enlightening.