PaulStoffregen/AltSoftSerial #31: Erros when compiling AltSoftSerial with TMRpcm

dinosaurfiles, 2016-09-25:
TMRpcm [link](https://github.com/TMRh20/TMRpcm)

Sample Code:
```
#include <TMRpcm.h>
TMRpcm tmrpcm; // TMRpcm Object

#include <AltSoftSerial.h>
AltSoftSerial BTserial; // AltSoftSerial BT Object
```

Renders a resulting compile error of:
```
"/home/<redacted>/tools/Arduino/hardware/tools/avr/bin/avr-gcc" -w -Os -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p  -o "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/v5.ino.elf" "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/sketch/v5.ino.cpp.o" "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/libraries/SD/File.cpp.o" "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/libraries/SD/SD.cpp.o" "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/libraries/SD/utility/Sd2Card.cpp.o" "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/libraries/SD/utility/SdFile.cpp.o" "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/libraries/SD/utility/SdVolume.cpp.o" "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/libraries/SPI/SPI.cpp.o" "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/libraries/TMRpcm-master/TMRpcm.cpp.o" "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/libraries/TMRpcm-master/pcmRF.cpp.o" "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/libraries/AltSoftSerial-1.4/AltSoftSerial.cpp.o" "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/core/core.a" "-L/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp" -lm
/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/libraries/AltSoftSerial-1.4/AltSoftSerial.cpp.o (symbol from plugin): In function `AltSoftSerial::read()':
(.text+0x0): multiple definition of `__vector_10'
/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/libraries/TMRpcm-master/TMRpcm.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
```

Full log:
```
Arduino: 1.6.10 (Linux), Board: "Arduino/Genuino Uno"

/home/<redacted>/tools/Arduino/arduino-builder -dump-prefs -logger=machine -hardware "/home/<redacted>/tools/Arduino/hardware" -tools "/home/<redacted>/tools/Arduino/tools-builder" -tools "/home/<redacted>/tools/Arduino/hardware/tools/avr" -built-in-libraries "/home/<redacted>/tools/Arduino/libraries" -libraries "/home/<redacted>/Arduino/libraries" -fqbn=arduino:avr:uno -ide-version=10610 -build-path "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/home/<redacted>/Desktop/ArduinoResources/v5/v5.ino"
/home/<redacted>/tools/Arduino/arduino-builder -compile -logger=machine -hardware "/home/<redacted>/tools/Arduino/hardware" -tools "/home/<redacted>/tools/Arduino/tools-builder" -tools "/home/<redacted>/tools/Arduino/hardware/tools/avr" -built-in-libraries "/home/<redacted>/tools/Arduino/libraries" -libraries "/home/<redacted>/Arduino/libraries" -fqbn=arduino:avr:uno -ide-version=10610 -build-path "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/home/<redacted>/Desktop/ArduinoResources/v5/v5.ino"
Using board 'uno' from platform in folder: /home/<redacted>/tools/Arduino/hardware/arduino/avr
Using core 'arduino' from platform in folder: /home/<redacted>/tools/Arduino/hardware/arduino/avr
Detecting libraries used...
"/home/<redacted>/tools/Arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/cores/arduino" "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/variants/standard" "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/sketch/v5.ino.cpp" -o "/dev/null"
"/home/<redacted>/tools/Arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/cores/arduino" "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/variants/standard" "-I/home/<redacted>/tools/Arduino/libraries/SD/src" "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/sketch/v5.ino.cpp" -o "/dev/null"
"/home/<redacted>/tools/Arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/cores/arduino" "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/variants/standard" "-I/home/<redacted>/tools/Arduino/libraries/SD/src" "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/libraries/SPI/src" "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/sketch/v5.ino.cpp" -o "/dev/null"
"/home/<redacted>/tools/Arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/cores/arduino" "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/variants/standard" "-I/home/<redacted>/tools/Arduino/libraries/SD/src" "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/libraries/SPI/src" "-I/home/<redacted>/Arduino/libraries/TMRpcm-master" "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/sketch/v5.ino.cpp" -o "/dev/null"
"/home/<redacted>/tools/Arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/cores/arduino" "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/variants/standard" "-I/home/<redacted>/tools/Arduino/libraries/SD/src" "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/libraries/SPI/src" "-I/home/<redacted>/Arduino/libraries/TMRpcm-master" "-I/home/<redacted>/Arduino/libraries/AltSoftSerial-1.4" "/tmp/build67c3bf7bf726ded6cf06e7e2c7f8d7ce.tmp/sketch/v5.ino.cpp" -o "/dev/null"
"/home/<redacted>/tools/Arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/cores/arduino" "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/variants/standard" "-I/home/<redacted>/tools/Arduino/libraries/SD/src" "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/libraries/SPI/src" "-I/home/<redacted>/Arduino/libraries/TMRpcm-master" "-I/home/<redacted>/Arduino/libraries/AltSoftSerial-1.4" "/home/<redacted>/tools/Arduino/libraries/SD/src/File.cpp" -o "/dev/null"
"/home/<redacted>/tools/Arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/cores/arduino" "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/variants/standard" "-I/home/<redacted>/tools/Arduino/libraries/SD/src" "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/libraries/SPI/src" "-I/home/<redacted>/Arduino/libraries/TMRpcm-master" "-I/home/<redacted>/Arduino/libraries/AltSoftSerial-1.4" "/home/<redacted>/tools/Arduino/libraries/SD/src/SD.cpp" -o "/dev/null"
"/home/<redacted>/tools/Arduino/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/cores/arduino" "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/variants/standard" "-I/home/<redacted>/tools/Arduino/libraries/SD/src" "-I/home/<redacted>/tools/Arduino/hardware/arduino/avr/libraries/SPI/src" "-I/home/<redacted>/Arduino/libraries/TMRpcm-master" "-I/home/<redacted>/Arduino/li [... cut ...]
collect2: error: ld returned 1 exit status
Using library SD at version 1.0.8 in folder: /home/<redacted>/tools/Arduino/libraries/SD 
Using library SPI at version 1.0 in folder: /home/<redacted>/tools/Arduino/hardware/arduino/avr/libraries/SPI 
Using library TMRpcm-master at version 1.0 in folder: /home/<redacted>/Arduino/libraries/TMRpcm-master 
Using library AltSoftSerial-1.4 at version 1.4 in folder: /home/<redacted>/Arduino/libraries/AltSoftSerial-1.4 
exit status 1
Error compiling for board Arduino/Genuino Uno.
```

PaulStoffregen, 2016-09-25:
uno has only one 16 bit timer.

closed 2016-09-25 by PaulStoffregen

dinosaurfiles, 2016-09-25:
Is there any way we could use the Bluetooth module for sending data (We used this tutorial http://www.martyncurrey.com/arduino-to-arduino-by-bluetooth/) and use a speaker to play an audio file at the same time?

PaulStoffregen, 2016-09-25:
Ask on Arduino's forum.  This issue tracker is for reporting software bugs, not for tech support.