PaulStoffregen/AltSoftSerial #32: Problem at 2400 baud (atmega328)

simonecab, 2016-11-01:
This library is a great work.
I need to use at the same time a GSM modem (on AltSoftSerial) and another device on SoftwareSerial that supports max 38Kbaud, thus I need to use the modem at 2400 baud.
The issue is: at 2400 baud in some situation (but sistematically) the first bytes of some TX packets are not transmitted. 
This appears to be influenced by the previous bytes transmitted. I do not have performance issues so I could perform any kind of reset.

PaulStoffregen, 2016-11-01:
Please check to make sure you have version 1.4.

If you want me to investigate, I need code and details to reproduce the problem.  I do not investigate issues until complete info is posted.

simonecab, 2016-11-01:
You are perfectly right! I understand so well. 
The version is 1.4. 
I'll try to reproduce the issue in a standalone manner (not easy due to custom hw!). I will try to reproduce the issue with a couple of atmega328, the second one with the HW serial at 2400baud doing a delayed echo.

If you have any hint (something like debug print or similar) please let me know!
Just for info, now I have an ORRIBLE workaround, I put 8 spaces before the AT commands... this makes things working (so far) 
:-)

simonecab, 2016-11-01:
I apologize! The problem is in the modem  as far as I can understand.

closed 2016-11-01 by simonecab