PaulStoffregen/AltSoftSerial #37: Using Digital Pin 10 on Arduino Uno

domdragon, 2017-06-01:
Hello. I use AltSoftSerial for my program in an Arduino Uno to read data from sensor circuits so I use D8 & D9. At the same time, I use a RH95 library that uses pin 10 as an SPI slave select for a LoRa shield (I checked its code and it uses digitalwrite to do this). The AltSoftSerial doesn't seem to respond properly, not getting data from the sensor circuits. I am also not able to send using the LoRa sheild. I thought pin 10 can be used normally with digitalwrite. Is there a problem with using pin 10 this way? Thanks.

amprodes, 2019-11-01:
Did you found a solution to this? I think that 10pin is for NSS, and you could use any pin on arduino

domdragon, 2019-11-01:
Sorry, I should have closed this a long time ago. I think I just used I2C for the sensors to resolve my issue.

closed 2019-11-01 by domdragon

dmak, 2020-01-20:
I have a similar challenge. As to my information, Ethernet Shield ENC28J60 uses pins D7-D13:
[image]
and that conflicts with default setting 8 (RX) and 9 (TX) for Arduino Nano. What is the walkaround?

PaulStoffregen, 2020-01-21:
AltSoftSerial pins are fixed.  There is no workaround, unless you use a different board.  All the wishful thinking and software trickery in the world will not magically make the timer1 hardware inside this AVR chip connect to different pins.  Atmel wired the timer ICP and compare features to those specific pins.  It is hard wired inside the chip.

PaulStoffregen, 2020-01-21:
Or you can use regular SoftwareSerial, with all it's problems blocking interrupts, but it does allow use of any pins because it uses the GPIO registers, rather than timer input capture and output compare.

dmak, 2020-01-21:
Thanks for this information, I appreciate. Probably Ethernet shield needs pin 8 for the same reason: it needs timer. In principal I need only to receive the information, hence only RX is needed. Is it possible to configure AltSoftSerial to use 9 for RX and 8 for TX (i.e. swap them)? Or there will be a conflict anyway (or side effect caused by timer)?