PaulStoffregen/AltSoftSerial #41: Arduino Robot

lukzmu, 2017-12-04:
I am wondering, is it possible to use AltSoftSerial in any way on the official Arduino Robot?
The OUTPUT_COMPARE_A_PIN is set to 5 on the AVR_ATmega32U4, and on the Arduino Robot it is connected to the buzzer, so can't access it in any normal way.

What could be solution to the problem?

PaulStoffregen, 2017-12-04:
Maybe edit AltSoftSerial_Boards.h to use the other timer?

lukzmu, 2017-12-04:
Wasn't sure if I can switch the pin numbers directly, not to do any damage.

closed 2018-11-09 by lukzmu

JaroslavMalik, 2019-02-23:
thanks https://www.youtube.com/watch?v=pCOm4lghRFg

fikrisaman, 2019-11-21:
Hello, what's the number of the pin that you changed to?

JaroslavMalik, 2020-04-22:
In AltSoftSerial_Boards.h :

// Arduino Leonardo & Yun (from Cristian Maglie)
//
#elif defined(ARDUINO_AVR_YUN) || defined(ARDUINO_AVR_LEONARDO) || defined(__AVR_ATmega32U4__)

  #define ALTSS_USE_TIMER1
  #define INPUT_CAPTURE_PIN		4  // receive
  #define OUTPUT_COMPARE_A_PIN	9 // transmit
  #define OUTPUT_COMPARE_B_PIN	10 // unusable PWM
  #define OUTPUT_COMPARE_C_PIN	11 // unusable PWM
/*
  #define ALTSS_USE_TIMER3
  #define INPUT_CAPTURE_PIN		13 // receive
  #define OUTPUT_COMPARE_A_PIN		5 // transmit
*/