PaulStoffregen/AltSoftSerial #45: Lost answers UART

githubrn, 2018-03-25:
### Description

sometimes the answers are lost uart. Module sim800L
AT+CLCC=1 (enable status call, 3 statuses: 1,0,2 1,0,0 1,0,6)

### Steps To Reproduce Problem

repeat the operation several times

### Hardware & Software

Board Arduino UNO
modules used SIM800L
Arduino IDE version 1.8.5
Operating system & version windows 7 x64

### Arduino Sketch

```cpp
#include <AltSoftSerial.h>

AltSoftSerial altSerial;
char c;
  
String resp="";

void setup() {
  Serial.begin(9600);
  //Serial.println("AltSoftSerial Test Begin");
  altSerial.begin(9600);
//  altSerial.println("at+cpowd=1");
//  delay(10000); 
  altSerial.println("at");
 // altSerial.println("atdxxxxxxxxxx;");
}
void loop() {
  if (altSerial.available()) {
    c = altSerial.read();
    Serial.print(c);   
  }
  
  if (Serial.available()) {
    c = Serial.read();
    altSerial.print(c);
    Serial.print(resp);
  }
 
 
}
```

### Errors or Incorrect Output

at <----- **manual input** 
OK <----**normal answer**
atdXXXXXXXXXX; <----- **manual input** 
OK <----**normal answer**

+CLCC: 1,0,2,0,0,"XXXXXXXXXX",129,""  <----**normal answer**
 atdXXXXXXXXXX;                 <----- **bad answer \0 (end line) and print next command, manual input** 
OK <----**normal answer**

+CLCC: 1,0,2,0,0,"XXXXXXXXXX",129,"" <----**normal answer**

+CLCC: 1,0,0,0,0,"XXXXXXXXXX",129,"" <----**normal answer**

+CLCC: 1,0,6,0,0,"XXXXXXXXXX",129,"" <----**normal answer**

BUSY <----**normal answer**
atdXXXXXXXXXX; <----- **manual input** 
OK <----**normal answer**

+CLCC: 1,0,2,0,0,"XXXXXXXXXX",129,"" <----**normal answer**

+CLCC: 1,0,6,0,0,"XXXXXXXXXX",129,"" <----**normal answer**

NO DIALTONE <----**normal answer**
atdXXXXXXXXXX; <----- **manual input** 
OK <----**normal answer**

+CLCC: 1,0,2,0,0,"XXXXXXXXXX",129,"" <----**normal answer**

+CLCC: 1,0,0,0,0,"XXXXXXXXXX",129,"" <----**normal answer**

+CLCC: 1,0,6,0,0,"XXXXXXXXXX",129,"" <----**normal answer**

BUSY <----**normal answer**
 
 _<-----cut more 10 try ----->_

atdXXXXXXXXXX; <----- **manual input** 
OK <----**normal answer**

+CLCC: 1,0,2,0,0,"XXXXXXXXXX",129,"" <----**normal answer**

atdXXXXXXXXXX; <----- **bad answer \0 (end line) and print next command, manual input** 
OK <----**normal answer**

+CLCC: 1,0,2,0,0,"XXXXXXXXXX",129,"" <----**normal answer**

+CLCC: 1,0,0,0,0,"XXXXXXXXXX",129,"" <----**normal answer**

+CLCC: 1,0,6,0,0,"XXXXXXXXXX",129,"" <----**normal answer**

BUSY <----**normal answer**

PaulStoffregen, 2018-03-27:
Is it possible to reproduce this problem using the SIM800L without a sim card installed?

In other words, if I buy a SIM800L but I do not have the sim card, can I connect it and get this result?  Or do I need to have a proper sim card for testing?

githubrn, 2018-03-27:
sim card is needed. Statuses are returned only when making a call.
SIM800L uses unsolicited notifications when reporting statuses.
I think a sim card can be any. I tried 2 operators. The result is the same.

How to use flow control altsoftserial?

manual:

`
1.5 Supported character sets
The SIM800 Series AT Command interface defaults to the IRA character set. The SIM800 Series
supports the following character sets:
GSM format
UCS2
HEX
IRA
PCCP
PCDN
8859-1
The character set can be set and interrogated using the "AT+CSCS" Command (3GPP TS
27.007). The character set is defined in GSM specification 3GPP TS 27.005.
The character set affects transmission and reception of SMS and SMS Cell Broadcast messages,
the entry and display of phone book entries text field and SIM Application Toolkit alpha strings.

1.6 Flow control
Flow control is very important for correct communication between the GSM engine and DTE.
For in the case such as a data or fax call, the sending device is transferring data faster than the
receiving side is ready to accept. When the receiving buffer reaches its capacity, the receiving
device should be capable to cause the sending device to pause until it catches up.
There are basically two approaches to achieve data flow control: software flow control and
hardware flow control. SIM800 Series support both two kinds of flow control.
In Multiplex mode, it is recommended to use the hardware flow control.

1.6.1 Software flow control (XON/XOFF flow control)
Software flow control sends different characters to stop (XOFF, decimal 19) and resume (XON,
decimal 17) data flow. It is quite useful in some applications that only use three wires on the serial interface.
The default flow control approach of SIM800 Series is hardware flow control (RTS/CTS flow
control), to enable software flow control in the DTE interface and within GSM engine, type the
following AT Command:
AT+IFC=1, 1
This setting is stored volatile, for use after restart, AT+IFC=1, 1 should be stored to the user
profile with AT&W.
NOTE:
The AT commands listed in the table of AT&W chapter should be stored to user profile with
AT&W for use after restart. Most other AT commands in V.25, 3GPP TS 27.005, 3GPP TS
27.007,GPRS will store parameters automatically and can be used after module restart.
Ensure that any communications software package (e.g. Hyper terminal) uses software flow
control.
NOTE:
Software Flow control should not be used for data calls where binary data will be transmitted or
received (e.g. TCP/IP) as the DTE interface may interpret binary data as flow control characters.

1.6.2 Hardware flow control (RTS/CTS flow control)
Hardware flow control achieves the data flow control by controlling the RTS/CTS line. When the
data transfer should be suspended, the CTS line is set inactive until the transfer from the
receiving buffer has completed. When the receiving buffer is ok to receive more data, CTS goes
active once again.
To achieve hardware flow control, ensure that the RTS/CTS lines are present on your application
platform.
`

PaulStoffregen, 2018-03-27:
Is there a way to reproduce this problem with other serial devices?  Maybe another Arduino board?  (I have plenty of those....)

githubrn, 2018-03-27:
this is possible. I will report the result.