PaulStoffregen/AltSoftSerial #46: Elimininate 'unused parameter' warnings

bperrybap, 2018-04-02:
Paul,
you can remove the unused parameter warnings in AltSoftSerial.h by doing this:
line 63:
```
AltSoftSerial(uint8_t rxPin, uint8_t txPin, bool inverse = false) { (void) rxPin; (void) txPin; (void) inverse;}
```

line 68
```
static void enable_timer0(bool enable) { (void) enable;}
```

JAY-M-ERICSSON, 2021-02-25:
it's probably better to add ` __attribute__ ((unused)) ` in front of all the infringing parameters

bperrybap, 2021-02-25:
I would suspect__attribute__ won't be as portable given it isn't part of the actual standard and is primarily a gcc thing.
While  Arduino and 3rd party developers currently seem to all be using gcc for their platform toolsets,  it might potentially be an issue at some point.

JAY-M-ERICSSON, 2021-02-25:
fair enough indeed - I had arduino in mind and I edited my own version of the library to add those. works fine but indeed a GCC thingy

bperrybap, 2021-02-25:
I think that given Arduino and its 3rd party platforms have used gcc from the beginning for well over 10 years now,  that there are probably many little gcc things scattered around so it likely is not an issue worth worrying over.
In fact, I've got a few gcc specific things in some of my libraries.

commit 8e86f89e (2021-02-26): Fix harmless compiler warnings on Arduino Uno
fixes #46
closed 2021-02-26 by PaulStoffregen