PaulStoffregen/AltSoftSerial #48: Conflict with Servo: Multiple definitions

flaghacker, 2018-04-30:
### Description

When creating both a `Servo` and a `AltSoftwareSerial` object I get a compiler error.

### Steps To Reproduce Problem

Make sure this library is in the `libraries` folder of the Arduino IDE, and try to compile the code below.

### Hardware & Software

Arduino UNO
Board: Arduino UNO
Arduino IDE version 1.8.5 (but also happens when compiling with PlatformIO)
Arduino AVR Boards 1.6.20
Windows 10

### Arduino Sketch

```cpp
#include <Arduino.h>
#include <AltSoftSerial.h>
#include <Servo.h>

AltSoftSerial bluetooth;
Servo servo;

void setup() {}
void loop() {}
```

### Errors or Incorrect Output

```
libraries\Servo\avr\Servo.cpp.o (symbol from plugin): In function `ServoCount':

(.text+0x0): multiple definition of `__vector_11'

libraries\AltSoftSerial\AltSoftSerial.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.
```

PaulStoffregen, 2018-04-30:
Yes, this is a well known conflict.  From the web page:

> AltSoftSerial uses a 16 bit hardware timer to generate the transmit output and measure the receive input waveforms. Any other library which needs the same timer, and the PWM pins which depend upon that timer, will not work.

flaghacker, 2018-04-30:
Ah that's unfortunate, I was hoping this library would allow me to use a Servo and something SoftwareSerial-like at the same time. Sorry for the dumb issue.

closed 2018-04-30 by flaghacker

davidbitton, 2018-05-20:
I'm seeing this issue as well with [SoftPWM](https://github.com/Palatis/arduino-softpwm). Is there a way I can "move" which timer is being used? I tried `#define ALTSS_USE_TIMER3` with little luck. Is there any workaround? I need to take receipt of serial data and output correlating PWM. Thanks.

davidbitton, 2018-05-20:
I think I "solved" my problem. I'm going to use a Pro Micro in lieu of a Pro Mini. The 32u4 proc will be causative to AltSoftSerial using Timer3; leaving Timer1 for SoftPWM. Thanks.

PaulStoffregen, 2018-05-22:
Yes, this is a well known conflict. From the web page:

> AltSoftSerial uses a 16 bit hardware timer to generate the transmit output and measure the receive input waveforms. Any other library which needs the same timer, and the PWM pins which depend upon that timer, will not work.

On boards using AVR ATMEGA328P, there is only one 16 bit timer.  It is a known limitation of this cheap hardware.  These libraries which require a 16 bit timer can not magically add more hardware timers.  Only you can do that, but using a better board with more capable hardware.

davidbitton, 2018-05-22:
After a bunch of sleuthing, I settled on that 32u4 proc which has an adequate number of timers. However one problem; not enough pins. I had to make a change in the boards.h file because the Micro Pro board doesn't have an accessible pin 13.