PaulStoffregen/AltSoftSerial #53: RX Buffer max size is 500byte?

dirkclemens, 2018-11-21:
When using the library with a Mega2560 board (rx pin 48) I always get a "serial buffer overload" error when 500 bytes were received.

```
    if (mySerial.available()) {
      buffer[idx] = mySerial.read();
      ...
      idx += 1;
  }

if (mySerial.overflow()) {
    Serial.print("SERIAL_OVERFLOW at: ");
    Serial.println(idx);
  }
```

I already changed the size to 
`#define RX_BUFFER_SIZE 1024`
but it still ends at 500 bytes read.

I read a stream with nearly 700 byte from my smart meter.

PaulStoffregen, 2018-11-21:
The head & tail variables are only 8 bits, so anything over 255 bytes could be expected to fail.

dirkclemens, 2018-11-22:
ok, thanks, but just setting them to uint16_t doesn't solve the problem, isn't it? (including all method internal variables)

Rothammel, 2019-02-15:
> ok, thanks, but just setting them to uint16_t doesn't solve the problem, isn't it? (including all method internal variables)

Dirk hast du eine Lösung gefunden? ich stehe vor dem selben Problem. mein Stromzähler erzeugt natürlich auch einen overflow.

dirkclemens, 2019-02-15:
@Rothammel unfortunately not, I stopped and went back to my Raspberry Pi / Python solution, which is running fine, but is somehow oversized ;-) Anyway, but it works.

Rothammel, 2019-02-15:
dann probiere einmal NeoSWSerial

closed 2025-11-17 by dirkclemens