PaulStoffregen/AltSoftSerial #57: Support Arduino Nano Every

harryberlin, 2019-06-30:
Is there a way to get it wokring with Arduino Nano Every?

PaulStoffregen, 2019-06-30:
Probably, but I'm not going to work on porting anytime soon.  Maybe you can find someone who knows enough about AVR timers to do this work and send a pull request?

KhoaAnhVo, 2020-09-10:
Hello I am working with Arduino Nano using AltSoftSerial.
I test the operation by sending data from PC -> Serial -> AltSoftSerial -> PC. And do the opposite.
AltSoftSerial reads data from PC very well. I have verified at 115200. But Write is not working correctly. I send "Serial Say hello!" -> AltSoftSerial -> PC result: "Seri? L Say hel? O!"
Please reconsider this issue. thanks.

harryberlin, 2020-09-10:
Note: Nano Every is not the same like usual Nano. Open your own issue. In my case, usual nano is working.