PaulStoffregen/AltSoftSerial #58: Support for Teensy 4.0

GLBaxter, 2019-11-10:
### Description
I am unable to compile any sketches that include <AltSoftSerial.h> against the Teensy 4.0 items I've purchased.

### Steps To Reproduce Problem
I am using Arduino IDE 1.8.10 and the latest Teensy Loader 1.48.  I can compile sketches (but not upload) for all the Teensy boards under the IDE Tools\Board:\Boards Manager up through Teensy 3.6.  In looking at the AltSoftSerial-master\config\AlstSoftSerial_Boards.h, Teensy 2.0, Teensy++2.0 and Teensy 3.x are listed (and other makes of boards), but nothing specifically w.r.t. a Teensy 4.0.

### Hardware & Software

Board  Teensy 4.0
Shields / modules used  None
Arduino IDE version 1.8.10
Teensyduino version (if using Teensy)  1.48
Version info & package name (from Tools > Boards > Board Manager)  Teensy 2.0 through Teensy 4.0
Operating system & version  Windows 10 Pro
Any other software or hardware?  None yet

NOTE:  I do have alternative AltSoftSerial installs, but they are all the current 1.4 version.

### Arduino Sketch

Verbatim: ---
#include <AltSoftSerial.h>

AltSoftSerial altSerial;

void setup() {
  Serial.begin(9600);
  Serial.println("AltSoftSerial Test Begin");
  altSerial.begin(9600);
  altSerial.println("Hello World");
}

void loop() {
  char c;

  if (Serial.available()) {
    c = Serial.read();
    altSerial.print(c);
  }
  if (altSerial.available()) {
    c = altSerial.read();
    Serial.print(c);
  }
}

### Errors or Incorrect Output
Arduino: 1.8.10 (Windows 10), TD: 1.48, Board: "Teensy 4.0, Serial, 600 MHz, Faster, US English"

C:\Program Files (x86)\Arduino_1_8_10\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino_1_8_10\hardware -hardware C:\Users\glsw0\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino_1_8_10\tools-builder -tools C:\Program Files (x86)\Arduino_1_8_10\hardware\tools\avr -tools C:\Users\glsw0\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino_1_8_10\libraries -libraries C:\Users\glsw0\OneDrive\Documents\Arduino\libraries -fqbn=teensy:avr:teensy40:usb=serial,speed=600,opt=o2std,keys=en-us -ide-version=10810 -build-path C:\Users\glsw0\AppData\Local\Temp\arduino_build_872522 -warnings=default -build-cache C:\Users\glsw0\AppData\Local\Temp\arduino_cache_74529 -verbose C:\Users\glsw0\OneDrive\Documents\Teensy\sketch_nov09a\sketch_nov09a.ino
C:\Program Files (x86)\Arduino_1_8_10\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino_1_8_10\hardware -hardware C:\Users\glsw0\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino_1_8_10\tools-builder -tools C:\Program Files (x86)\Arduino_1_8_10\hardware\tools\avr -tools C:\Users\glsw0\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino_1_8_10\libraries -libraries C:\Users\glsw0\OneDrive\Documents\Arduino\libraries -fqbn=teensy:avr:teensy40:usb=serial,speed=600,opt=o2std,keys=en-us -ide-version=10810 -build-path C:\Users\glsw0\AppData\Local\Temp\arduino_build_872522 -warnings=default -build-cache C:\Users\glsw0\AppData\Local\Temp\arduino_cache_74529 -verbose C:\Users\glsw0\OneDrive\Documents\Teensy\sketch_nov09a\sketch_nov09a.ino
Using board 'teensy40' from platform in folder: C:\Program
Using core 'teensy4' from platform in folder: C:\Program
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino_1_8_10\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=148 -DARDUINO=10810 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino_1_8_10\\hardware\\teensy\\avr\\cores\\teensy4" "C:\\Users\\glsw0\\AppData\\Local\\Temp\\arduino_build_872522\\sketch\\sketch_nov09a.ino.cpp" -o nul
Alternatives for AltSoftSerial.h: [AltSoftSerial@1.4]
ResolveLibrary(AltSoftSerial.h)
  -> candidates: [AltSoftSerial@1.4]
"C:\\Program Files (x86)\\Arduino_1_8_10\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=148 -DARDUINO=10810 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino_1_8_10\\hardware\\teensy\\avr\\cores\\teensy4" "-IC:\\Users\\glsw0\\OneDrive\\Documents\\Arduino\\libraries\\AltSoftSerial" "C:\\Users\\glsw0\\AppData\\Local\\Temp\\arduino_build_872522\\sketch\\sketch_nov09a.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino_1_8_10\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=148 -DARDUINO=10810 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino_1_8_10\\hardware\\teensy\\avr\\cores\\teensy4" "-IC:\\Users\\glsw0\\OneDrive\\Documents\\Arduino\\libraries\\AltSoftSerial" "C:\\Users\\glsw0\\OneDrive\\Documents\\Arduino\\libraries\\AltSoftSerial\\AltSoftSerial.cpp" -o nul
Error while detecting libraries included by C:\Users\glsw0\OneDrive\Documents\Arduino\libraries\AltSoftSerial\AltSoftSerial.cpp
Generating function prototypes...
"C:\\Program Files (x86)\\Arduino_1_8_10\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=148 -DARDUINO=10810 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino_1_8_10\\hardware\\teensy\\avr\\cores\\teensy4" "-IC:\\Users\\glsw0\\OneDrive\\Documents\\Arduino\\libraries\\AltSoftSerial" "C:\\Users\\glsw0\\AppData\\Local\\Temp\\arduino_build_872522\\sketch\\sketch_nov09a.ino.cpp" -o "C:\\Users\\glsw0\\AppData\\Local\\Temp\\arduino_build_872522\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Program Files (x86)\\Arduino_1_8_10\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\glsw0\\AppData\\Local\\Temp\\arduino_build_872522\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Program Files (x86)\\Arduino_1_8_10\\hardware\\teensy/../tools/precompile_helper" "C:\\Program Files (x86)\\Arduino_1_8_10\\hardware\\teensy\\avr/cores/teensy4" "C:\\Users\\glsw0\\AppData\\Local\\Temp\\arduino_build_872522" "C:\\Program Files (x86)\\Arduino_1_8_10\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -x c++-header -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=148 -DARDUINO=10810 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino_1_8_10\\hardware\\teensy\\avr/cores/teensy4" "C:\\Users\\glsw0\\AppData\\Local\\Temp\\arduino_build_872522/pch/Arduino.h" -o "C:\\Users\\glsw0\\AppData\\Local\\Temp\\arduino_build_872522/pch/Arduino.h.gch"
"C:\\Program Files (x86)\\Arduino_1_8_10\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -c -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=148 -DARDUINO=10810 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Users\\glsw0\\AppData\\Local\\Temp\\arduino_build_872522/pch" "-IC:\\Program Files (x86)\\Arduino_1_8_10\\hardware\\teensy\\avr\\cores\\teensy4" "-IC:\\Users\\glsw0\\OneDrive\\Documents\\Arduino\\libraries\\AltSoftSerial" "C:\\Users\\glsw0\\AppData\\Local\\Temp\\arduino_build_872522\\sketch\\sketch_nov09a.ino.cpp" -o "C:\\Users\\glsw0\\AppData\\Local\\Temp\\arduino_build_872522\\sketch\\sketch_n [... cut ...]
C:\Users\glsw0\OneDrive\Documents\Arduino\libraries\AltSoftSerial/AltSoftSerial.h:47:43: note: in expansion of macro 'ALTSS_BASE_FREQ'

  static void begin(uint32_t baud) { init((ALTSS_BASE_FREQ + baud / 2) / baud); }

                                           ^

Multiple libraries were found for "AltSoftSerial.h"
 Used: C:\Users\glsw0\OneDrive\Documents\Arduino\libraries\AltSoftSerial
Using library AltSoftSerial at version 1.4 in folder: C:\Users\glsw0\OneDrive\Documents\Arduino\libraries\AltSoftSerial 
Error compiling for board Teensy 4.0.

Regards.

JM-F, 2020-06-10:
Having the same problem here. Using Arduino 1.8.11 with Teensy 4.0.
Any idea?

dunland, 2020-11-24:
We've got the same problem with the recently purchased teensy 4.1
That's a big problem, since I can't use the sparkfun Tsunami device anymore!

I get the `"Please define your board timer and pins"` Error from AltSoftSerial_Boards.h.. 
I think adding a definition for teensy 4.1 here might be a first step.. But I don't how really, and I can't anticipate the consequences!