PaulStoffregen/AltSoftSerial #60: Implement Inverse flag to invert sense of RX and TX like SoftwareSerial

fourstix, 2020-06-06:
As a user I need to reverse the sense of the RX and TX lines to communicate with a particular piece of hardware as in SoftwareSerial, but I also need the full-duplex communication support of AltSoftSerial.

### Description

Please see Pull Request #59  

New feature.  Add the Inverse flag to AltSoftSerial Library with appropriate constructors.  In the Transmission and Reception code, invert the outputs with the flag is true.  When the flag is the default value of false, the code functions exactly as before.  The original constructors are unchanged.

### Steps To Reproduce Problem

New Feature.

### Hardware & Software

Tested with an Sparkfun RedBoard Arduino
Arduino IDE version 1.18.10
Teensyduino version (if using Teensy)
1802 Membership Card running MCSMP20J ROM (with inverse logic on the RX and TX lines)
and for regression testing
1802 Membership Card running MCPSMP20A ROM (with positive logic on RX and TX lines)

### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)
#include <Arduino.h>
/*
  Serial Connections for 1802 Membership Card MCSMP20J

  Arduino Pin     MCard1802
    9 (TX)        /EF3 (RX)   P1 - 27 
    8 (RX)          Q  (TX)   P1 - 12
    GND             GND       P1 -  1 (or P1 - 30)
*/

//Note: SoftwareSerial is *NOT* Full-duplex and won't work here.
#include <AltSoftSerial.h>

//RX pin 8, TX pin 9, with RX and TX inversion
AltSoftSerial altSerial(true);

void setup() {
 //Initialize serial for input and debugging
  Serial.begin(115200);
 //Initialize AltSoftSerial for communication to 1802
  altSerial.begin(1200);
  Serial.println("Ready!");
  //Establish communication with <CR>
  altSerial.write('\r'); 
}

void loop() {
  // If anything comes from Arduino IDE
  if (Serial.available()) {      
    // get the new byte:
    char nextChar = (char)Serial.read();            
    altSerial.write(nextChar);     
   //wait a bit for 1802 to process character sent
   delay(20);
  }
  if (altSerial.available()) {
    // get the new byte:
    char inChar = (char) altSerial.read();

    //Mask off high bit
    inChar = (inChar & 0x7F);
   //Write to Arduino Serial Monitor
   Serial.write(inChar);
  } // if altSerial.available
}
```

I have submitted pull request #59 to add the support for the inverse argument to the AltSoftSerial library.  Merging Pull Request #59 will close this issue.

cimba007, 2021-03-27:
This library is a bad joke:

```
// for drop-in compatibility with NewSoftSerial, rxPin & txPin ignored
AltSoftSerial(uint8_t rxPin, uint8_t txPin, bool inverse = false) { }
```

pretending to be compatible when in fact it is not

fourstix, 2021-07-14:
That constructor behavior is in the original library I only addressed the inverse flag in this Pull Request, which I needed for my own use.  One can set rxPin and txPin in this constructor but they are ignored.

Frankly, I don't like the dummy constructor either, but this PR leaves it up to the maintainer. At this point, though, it doesn't look like the library is being maintained anymore.   So this is probably a moot issue.

I plan to just remove this constructor from my own fork of this code and maintain the code there.