PaulStoffregen/AltSoftSerial #61: Couldnt transmit correct data using AltSoftSerial

Adee-CV, 2021-03-24:
Describe your problem: 
Not transmitting correct from AltSoftSerial object to Arduino Serial.

### Steps To Reproduce Problem

### Hardware & Software

Board : Arduino UNO
Arduino IDE version : 1.8.13
Version info & package name (from Tools > Boards > Board Manager)
Operating system & version : WIN 10 Pro
Any other software or hardwa
[echo.txt](https://github.com/PaulStoffregen/AltSoftSerial/files/6195662/echo.txt)
re? : Using  "echo" example with serial ports..... 

### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)
#include <Arduino.h>

 libraries: AltSoftSerial

void setup() {
}

void loop() {
}
```

### Errors or Incorrect Output

From Arduino Serial object to  AltSoftSerial data transmits perfectly.
From AltSoftSerial to Arduino serial only data with odd ASCII values transmits  correctly and even , its wrong. e.g. if transfers character 'A' its shows  'A' but  when i transmits 'B' it wont. Same 1,3,5,7,9 transmits correctly while 0,2,4,6,8 wont

PaulStoffregen, 2021-03-30:
A complete program must be shown to reproduce this problem.  The hardware connection used must also be provided.  If more than 1 board is used, the complete program for each board is needed.

Please post the necessary info to reproduce this problem.

PaulStoffregen, 2021-03-30:
Ok, I see echo.txt is this code:

```
#include <AltSoftSerial.h>

// AltSoftSerial always uses these pins:
//
// Board          Transmit  Receive   PWM Unusable
// -----          --------  -------   ------------
// Teensy 3.0 & 3.1  21        20         22
// Teensy 2.0         9        10       (none)
// Teensy++ 2.0      25         4       26, 27
// Arduino Uno        9         8         10
// Arduino Leonardo   5        13       (none)
// Arduino Mega      46        48       44, 45
// Wiring-S           5         6          4
// Sanguino          13        14         12

AltSoftSerial altSerial;

void setup() {
  Serial.begin(9600);
  while (!Serial) ; // wait for Arduino Serial Monitor to open

  altSerial.begin(9600);

}

void loop() {
  char c;

  if (Serial.available()) {
    c = Serial.read();

  }
  if (altSerial.available()) {
    c = altSerial.read();
    Serial.print(c);
  }
}
```

Please give the precise details on hardware connection.  Is a wire connected to pin 8?  I need you to say **exactly** how you have the hardware configured while running this program.  A diagram or photo is best.

Are you using Arduino Uno?  It is a genuine Uno, or a clone / counterfeit?  Details matter!

I also need to know exactly the procedure to perform the test.  This code does not transmit any data!  Maybe you are typing in the Arduino Serial Monitor.  Maybe you have some other hardware connected?  I can not see your computer, I can not see your workbench.  I need you to give the exact details so I can do the same thing you are doing to reproduce the problem.

Adee-CV, 2021-03-30:
thanks, i checked with diff hardware and found working....... Thanks for updates

closed 2021-03-30 by Adee-CV