PaulStoffregen/AltSoftSerial #65: Add Support for ATMEGA328PB Please

maesoph, 2022-03-06:
### Description

Describe your problem.

Does not appear to have support for ATMEGA328PB.  

### Steps To Reproduce Problem

Please give detailed instructions needed for anyone to attempt to reproduce the problem.

Just try compiling for ATMEGA328PB

### Hardware & Software

ATMEGA328PB

### Errors or Incorrect Output
Error while detecting libraries included by C:\Users\maeso\Documents\Arduino\libraries\AltSoftSerial\AltSoftSerial.cpp

Matejisko, 2023-01-04:
Hi, i had  same problem...

Using MCUdude/MiniCore witch 328PB variant selected i was unable to compile the program with AltSoftSerial library
I got many errors like:

_AltSoftSerial_Boards.h: 136:2: error: #error "Please define your board timer and pins"_
   _#error "Please define your board timer and pins"_

_AltSoftSerial.cpp: 74:3: error: 'CONFIG_TIMER_NOPRESCALE' was not declared in this scope_
   _CONFIG_TIMER_NOPRESCALE()_
.
.
.

But solution seems to be very easy. In file ...**\libraries\AltSoftSerial\config\AltSoftSerial_Boards.h** 
change line (85) : 
`#elif defined(__AVR_ATmega168__) || defined(__AVR_ATmega328P__)`

to:

`#elif defined(__AVR_ATmega168__) || defined(__AVR_ATmega328P__) || defined(__AVR_ATmega328PB__)`

Save the file, restart arduino and try.

Hope it helps

PaulStoffregen, 2023-01-04:
@Matejisko - Do you know what is different with the "PB" chip?  Please confirm the code did work when you ran it on a real PB chip and I will put this change into the code.

jacek-rozen, 2024-11-24:
Hi @PaulStoffregen ,

I hit the same issue and would suggest to add this fix with this single add on. 

First of all, ATMEGA328PB is a next release of ATMEGA328P. Manufacturer writes that: "_The code that is available for your existing ATmega328 variants will continue to work on the new ATmega328PB device._". 

Second of all, ATMEGA328 and ATMEGA328P are not recommended for new designs (https://www.microchip.com/en-us/application-notes/an42559), hence ATMEGA328PB should be used, hence libraries should adjust.

Third of all, detailed differencies are described here: [https://ww1.microchip.com/downloads/aemDocuments/documents/OTH/ApplicationNotes/ApplicationNotes/Atmel-42559-Differences-between-ATmega328P-and-ATmega328PB_ApplicationNote_AT15007.pdf](https://ww1.microchip.com/downloads/aemDocuments/documents/OTH/ApplicationNotes/ApplicationNotes/Atmel-42559-Differences-between-ATmega328P-and-ATmega328PB_ApplicationNote_AT15007.pdf)

In nutshell - there are more pins, and one GND pin changed its function. But readers - please check details for your designs and do not rely on this simplified description.

Given all above, I suggest to change.

commit 62a768a1 (2024-11-27): Support ATmega328PB, fixes #65