PaulStoffregen/AltSoftSerial #69: Add support for renesas_uno architecture

kalyalien, 2023-08-26:
### Description

Describe your problem.
 
ERROR MESSAGE WITH UNO R4 WIFI

WARNING: library AltSoftSerial claims to run on avr architecture(s) and may be incompatible with your current board which runs on renesas_uno architecture(s).
In file included from C:\Users\ClinicalUser\Documents\Arduino\libraries\WiFiEsp\src/WiFiEsp.h:30:0,
                 from C:\Users\ClinicalUser\Desktop\Whats The Time Mr Wolf\WhatsTheTimeMrWolf\WhatsTheTimeMrWolf.ino:45:
C:\Users\ClinicalUser\Documents\Arduino\libraries\WiFiEsp\src/utility/EspDrv.h:315:9: error: reference to 'RingBuffer' is ambiguous
  static RingBuffer ringBuf;
         ^~~~~~~~~~
In file included from C:\Users\ClinicalUser\Documents\Arduino\libraries\WiFiEsp\src/utility/EspDrv.h:26:0,
                 from C:\Users\ClinicalUser\Documents\Arduino\libraries\WiFiEsp\src/WiFiEsp.h:30,
                 from C:\Users\ClinicalUser\Desktop\Whats The Time Mr Wolf\WhatsTheTimeMrWolf\WhatsTheTimeMrWolf.ino:45:
C:\Users\ClinicalUser\Documents\Arduino\libraries\WiFiEsp\src/utility/RingBuffer.h:23:7: note: candidates are: class RingBuffer
 class RingBuffer
       ^~~~~~~~~~
In file included from C:\Users\ClinicalUser\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino/SafeRingBuffer.h:24:0,
                 from C:\Users\ClinicalUser\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino/Serial.h:38,
                 from C:\Users\ClinicalUser\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino/Arduino.h:18,
                 from C:\Users\ClinicalUser\AppData\Local\Temp\arduino\sketches\BEBAE33B41EEEF4B5D65B3C16CDA8391\sketch\WhatsTheTimeMrWolf.ino.cpp:1:
C:\Users\ClinicalUser\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino/api/RingBuffer.h:59:41: note:                 typedef class arduino::RingBufferN<64> arduino::RingBuffer
 typedef RingBufferN<SERIAL_BUFFER_SIZE> RingBuffer;
                                         ^~~~~~~~~~
In file included from C:\Users\ClinicalUser\AppData\Local\Temp\arduino\sketches\BEBAE33B41EEEF4B5D65B3C16CDA8391\sketch\WhatsTheTimeMrWolf.ino.cpp:1:0:
C:\Users\ClinicalUser\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino/Arduino.h:115:17: error: conflicting declaration 'AltSoftSerial _UART2_'
 #define Serial1 _UART2_
                 ^
C:\Users\ClinicalUser\Desktop\Whats The Time Mr Wolf\WhatsTheTimeMrWolf\WhatsTheTimeMrWolf.ino:52:17: note: in expansion of macro 'Serial1'
   AltSoftSerial Serial1;
                 ^~~~~~~
In file included from C:\Users\ClinicalUser\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino/Arduino.h:18:0,
                 from C:\Users\ClinicalUser\AppData\Local\Temp\arduino\sketches\BEBAE33B41EEEF4B5D65B3C16CDA8391\sketch\WhatsTheTimeMrWolf.ino.cpp:1:
C:\Users\ClinicalUser\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino/Serial.h:104:13: note: previous declaration as 'UART _UART2_'
 extern UART _UART2_;
             ^~~~~~~

exit status 1

Compilation error: exit status 1

### Steps To Reproduce Problem

Add compatibility

### Hardware & Software

Board Arduino Uno R4 WIFI
Shields / modules used
Arduino IDE version
Teensyduino version (if using Teensy)
Version info & package name (from Tools > Boards > Board Manager)
Operating system & version
Any other software or hardware?

### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)
#include <Arduino.h>

// libraries: give links/details so anyone can compile your code for the same result

void setup() {
}

void loop() {
}
```

### Errors or Incorrect Output

WARNING: library AltSoftSerial claims to run on avr architecture(s) and may be incompatible with your current board which runs on renesas_uno architecture(s).
In file included from C:\Users\ClinicalUser\Documents\Arduino\libraries\WiFiEsp\src/WiFiEsp.h:30:0,
                 from C:\Users\ClinicalUser\Desktop\Whats The Time Mr Wolf\WhatsTheTimeMrWolf\WhatsTheTimeMrWolf.ino:45:
C:\Users\ClinicalUser\Documents\Arduino\libraries\WiFiEsp\src/utility/EspDrv.h:315:9: error: reference to 'RingBuffer' is ambiguous
  static RingBuffer ringBuf;
         ^~~~~~~~~~
In file included from C:\Users\ClinicalUser\Documents\Arduino\libraries\WiFiEsp\src/utility/EspDrv.h:26:0,
                 from C:\Users\ClinicalUser\Documents\Arduino\libraries\WiFiEsp\src/WiFiEsp.h:30,
                 from C:\Users\ClinicalUser\Desktop\Whats The Time Mr Wolf\WhatsTheTimeMrWolf\WhatsTheTimeMrWolf.ino:45:
C:\Users\ClinicalUser\Documents\Arduino\libraries\WiFiEsp\src/utility/RingBuffer.h:23:7: note: candidates are: class RingBuffer
 class RingBuffer
       ^~~~~~~~~~
In file included from C:\Users\ClinicalUser\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino/SafeRingBuffer.h:24:0,
                 from C:\Users\ClinicalUser\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino/Serial.h:38,
                 from C:\Users\ClinicalUser\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino/Arduino.h:18,
                 from C:\Users\ClinicalUser\AppData\Local\Temp\arduino\sketches\BEBAE33B41EEEF4B5D65B3C16CDA8391\sketch\WhatsTheTimeMrWolf.ino.cpp:1:
C:\Users\ClinicalUser\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino/api/RingBuffer.h:59:41: note:                 typedef class arduino::RingBufferN<64> arduino::RingBuffer
 typedef RingBufferN<SERIAL_BUFFER_SIZE> RingBuffer;
                                         ^~~~~~~~~~
In file included from C:\Users\ClinicalUser\AppData\Local\Temp\arduino\sketches\BEBAE33B41EEEF4B5D65B3C16CDA8391\sketch\WhatsTheTimeMrWolf.ino.cpp:1:0:
C:\Users\ClinicalUser\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino/Arduino.h:115:17: error: conflicting declaration 'AltSoftSerial _UART2_'
 #define Serial1 _UART2_
                 ^
C:\Users\ClinicalUser\Desktop\Whats The Time Mr Wolf\WhatsTheTimeMrWolf\WhatsTheTimeMrWolf.ino:52:17: note: in expansion of macro 'Serial1'
   AltSoftSerial Serial1;
                 ^~~~~~~
In file included from C:\Users\ClinicalUser\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino/Arduino.h:18:0,
                 from C:\Users\ClinicalUser\AppData\Local\Temp\arduino\sketches\BEBAE33B41EEEF4B5D65B3C16CDA8391\sketch\WhatsTheTimeMrWolf.ino.cpp:1:
C:\Users\ClinicalUser\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2\cores\arduino/Serial.h:104:13: note: previous declaration as 'UART _UART2_'
 extern UART _UART2_;
             ^~~~~~~

exit status 1

Compilation error: exit status 1

PaulStoffregen, 2023-08-26:
This work needs to be done by a software developer with deep knowledge of the Uno R4 hardware timers.

kalyalien, 2023-08-27:
Hi PaulThanks for letting me know* Taskmaster Season 9 (Channel 4/Netflix)* The Ballad of Renegade Nell (Disney+)* Back To The Future (Adelphi Theatre)* Time Travellers Wife (West End, Apollo Theatre)* Represented by Richard Kort ManagementKalyanT: 07834547351 | W:garethkalyan.com